MoonshineSG/Octoprint-Filament #1: /dev/mem access issue on Octopi

lobermann, 2016-06-26:
Hi there,

just started using your plugin, very awesome I have to say! Didn't get it running in Marlin directly, but this is a much nicer approach.

Anyway, I want to share with you a small issue I encountered. The problem is with the access rights on the Pi for /dev/mem. As you describe it correctly you have to change the permissions, but this still does not allow you to access the device on the Pi.

```
2016-06-26 09:51:48,058 - octoprint.plugin - ERROR - Error while calling plugin octoprint_filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.13-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/.octoprint/plugins/octoprint_filament/__init__.py", line 36, in on_after_startup
    GPIO.setup(self.PIN_FILAMENT, GPIO.IN)
RuntimeError: No access to /dev/mem.  Try running as root!
```

Where the device is having the correct permissions
```
pi@octopi ~/.octoprint/logs $ ls -la /dev/mem
crwxrwxrwx 1 root kmem 1, 1 Jan  1  1970 /dev/mem
``` 

Only way I got it running was by starting Octoprint as root, by modifying following lines in the the /etc/default/octoprint:
```
# The init.d script will only run if this variable non-empty.
OCTOPRINT_USER=root

# What arguments to pass to octoprint, usually no need to touch this
DAEMON_ARGS="--host=$HOST --port=$PORT --iknowwhatimdoing"
```

and then copying the .octoprint folder from /home/pi to /root , so that all my stuff is still there.

Just want to let you know that,  not sure if there is a workaround for that, I couldn't find one in the short time I searched.

Cheers,
Lukas

MoonshineSG, 2016-06-26:
Strange.... And you have  RPi.GPIO version greater than 0.6.0 ?

lobermann, 2016-06-26:
Yes, version is greater.

```
pi@octopi ~ $ python
Python 2.7.3 (default, Mar 18 2014, 05:13:23) 
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO 
>>> print GPIO.VERSION
0.6.2
```

MoonshineSG, 2016-06-27:
Sorry, I was reading this on a smartphone and I didn't noticed the obvious:

chmod a+rw /dev/**gpio**mem

lobermann, 2016-06-27:
Yes, but there is no such device on my octopi. And also the plugin wants to access /dev/mem instead of /dev/gpiomem . Maybe there are differences between versions? My Octopi, the underlaying system, is at least a year old already, maybe a change there?

MoonshineSG, 2016-06-27:
maybe... 

can you do a `uname -a` ?

I'm running quite a new version as I am on RPi3 with jessie ("4.1.21-v7+ #872 SMP Wed Apr 6 17:34:14 BST 2016 armv7l GNU/Linux"

I'm not sure whats worse.. running OctoPrint as root or attempting an upgrade and possible complications. A backup SD card would be advisable in case you decide to upgrade.

lobermann, 2016-06-28:
Runnin a bit older version:
```Linux octopi 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux```

I will organize a new SD card and try the latest octopi, see if there is a difference.

closed 2016-07-24 by MoonshineSG