MoonshineSG/Octoprint-Filament #7: API location no API key found

brotherchris, 2016-08-26:
When I try to access the API location /plugin/filament/status. I only get the message. No API key provided. How would you provide it the octoprint key?

MoonshineSG, 2016-08-26:
See http://docs.octoprint.org/en/master/api/general.html

MoonshineSG, 2016-08-27:
The plugin will pause the print when the filament runs out. The additional API is provided and it can be called from other plugins which might need to know the status of the filament. It would be called as described in the above mentioned document.

> The API key must be supplied in the custom HTTP header X-Api-Key, e.g.
> 
> GET /plugin/filament/status HTTP/1.1
> Host: example.com
> X-Api-Key: abcdef...
> 

(A better implementation would be an event base trigger with a subscription model, but thats a different story.)

closed 2019-05-29 by brotherchris