MoonshineSG/Octoprint-Filament #8: Error while calling plugin filament

Werja, 2016-09-01:
from octoprint Log:

2016-09-01 15:53:54,607 - octoprint.plugins.filament - INFO - Running RPi.GPIO version '0.6.2'...
2016-09-01 15:53:54,610 - octoprint.plugins.filament - INFO - Filament Sensor Plugin [filament] initialized...
...
2016-09-01 15:53:57,605 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup on GPIO [2]...
...
2016-09-01 16:36:19,306 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 56, in on_event
    self.setup_gpio()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 70, in setup_gpio
    GPIO.add_event_detect(self.PIN_FILAMENT, GPIO.FALLING, callback=self.check_gpio, bouncetime=self.BOUNCE)
RuntimeError: Failed to add edge detection

MoonshineSG, 2016-09-02:
what pin number are you using ?

Werja, 2016-09-02:
2016-09-01 15:53:57,605 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup on GPIO [2]...
The status request via API works!

MoonshineSG, 2016-09-02:
That is odd.... 
Just trying to see what can be wrong... Did you do a "chmod a+rw /dev/gpiomem" ? Assume running under 'pi' user ?  What OS version ?

Werja, 2016-09-02:
OS Version 7 (wheezy) 
running under pi user
I have the "chmod a+rw /dev/gpiomem" added to the octoprint startscript

MoonshineSG, 2016-09-02:
can you run a "sudo chmod a+rw /dev/gpiomem" from terminal ? just run it once and restart OctoPrint .

I don't think you need that to be in the startup script...

MoonshineSG, 2016-09-02:
Then again... if you said the request via API works it has nothing to do with permissions...

Werja, 2016-09-02:
yes..
{
  "status": "1"
}
when filament is detected and
{
  "status": "0"
}
if not detected

MoonshineSG, 2016-09-02:
I which I can say I know how to help... 
what's your hardware like ? RPi2 RPi3 ? 
I can try to do an exact setup and see what happens, but it might take a while for me to have the time to do that... 

Not much help from google when searching for "Failed to add edge detection"...

Werja, 2016-09-02:
ok, many thanks for your help.
I think it is a problem in my individual RPi / Octoprint configuration.
I will give you feedback, when the problem is solve.

MoonshineSG, 2016-09-02:
Please do. It might help in the future. 
Hope you solve it!

ghvader, 2016-10-23:
I have the same exact issue as the OP.
any suggestions?

so the RPI and octoprint see's the change in filament state. but it never pauses.
from the log
setup
2016-10-23 18:17:02,127 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup on GPIO [24]...
then when called to act
2016-10-23 18:17:46,314 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2016-10-23 18:17:47,354 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 56, in on_event
    self.setup_gpio()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 70, in setup_gpio
    GPIO.add_event_detect(self.PIN_FILAMENT, GPIO.FALLING, callback=self.check_gpio, bouncetime=self.BOUNCE)
RuntimeError: Failed to add edge detection

MoonshineSG, 2016-10-24:
The "RuntimeError: Failed to add edge detection" seams to be a problem with the underlying OS.

ghvader, 2016-10-25:
i was using the octopi image, today i went ahead and started fresh with raspbian and installed octoprint manually. i was able to get the plugin working correctly. 
it did indeed pause the print..
however i have a follow up question is it possible to turn off the heated bed and extruders when it's paused?
say you leave the house or print over night and go to bed and the filiment breaks, 
the printer gets paused, but your gear stays @ 200c+ the whole time. could be hours..
doesn't seem safe to me.  is there a way to send a M command(s) along with the pause?

MoonshineSG, 2016-10-25:
@ghvader if you install the devel branch, the plugin will fire `FILAMENT_RUNOUT` event. You can then, write your own plugin to handle that event in any way you choose. 

A temporary solution, is to write the M commands to turn off the heater and bed in the OctoPrint Settings "GCODE Scripts" / "After print job is paused"  (just don't forget to turn it off otherwise ANY pause will stop your heaters!!!)

Btw, you should never leave the house with the printer running.

ghvader, 2016-10-25:
I saw those options just after i posted my comment.. so i came back for an edit.
I'll give the M codes a try.

yeah yeah :)

MoonshineSG, 2016-10-25:
@ghvader Wait a couple of minutes and I'm updating the devel. I'm changing "pause" to "gcode" and you can write whatever you want there... This should work for you