MoonshineSG/Octoprint-Filament #9: Logging

cbusillo, 2016-09-01:
Hi, great plugin.  I am really excited to use it.

The API returns status -1.  I cannot figure out what I am doing wrong.  From bash I am able to read if the sensor is tripped or not tripped.  The plugin installed correctly.

I cannot seem to find the log file.  I see in the code you have good logging.  It doesn't show up in the main log, and a log for the plugin is not generated.

Hopefully you can point me in the right direction.  I'm sure its a dependency thing.

MoonshineSG, 2016-09-02:
I assume you added the settings in `config.yaml` ? Which pin are you using ? What version of OctoPrint ? 

All the logs can be found under `/home/pi/.octoprint/logs/octoprint.log`. If you can't see anything check for failed loading of the plugin.

cbusillo, 2016-09-02:
I added the settings in the config.yaml.  I am using GPIO 2 (hardware pin 3).  Octo Print version 1.2.15.

Plugin shows loaded in the log.  When I access the API I get status -1.  There is nothing added in the log.

MoonshineSG, 2016-09-02:
the setting in config.yaml  is under plugins..., right ?

The logs show the message "Filament Sensor Plugin setup on GPIO 2..."

cbusillo, 2016-09-02:
Nope!!  Thanks so much.  I see you already updated the documentation.

closed 2016-09-02 by cbusillo