MoonshineSG/Octoprint-Filament #10: normally closed switch

cbusillo, 2016-09-02:
My switch is a normally closed switch.  When polling the API it shows 0 when the filament is loaded and 1 when it is not loaded.  Is there a variable I can set for a normally closed switch?

MoonshineSG, 2016-09-02:
At the moment there isn't a way to invert the behaviour. Let me think over the weekend and I'll get something done.

If you need smth fast, you can change the code so that the api returns reverse values and instead of  FALLING it should check for RASING.

cbusillo, 2016-09-02:
Thanks for the quick response!  I will keep an eye out for your next release!  Maybe a variable that says up or down?  If I were to simply changed the installed code do I have to recompile?  I haven't used Python in a long time.

MoonshineSG, 2016-09-03:
no, you don't need to recompile but you need to restart octoprint so it's loads the changed code

cbusillo, 2016-09-03:
I am able to get the correct status returned from the API by swapping the status var.  Do you have a good suggestion for code to pause the printer?

MoonshineSG, 2016-09-13:
please try the new codes that allow both type of switches. set `filament` to reflect your system

(codes not yet as a release, so you need to deploy manually)

cbusillo, 2016-09-14:
Super awesome.  It works!!!  My printer is pausing randomly though.  Does the script set a software type pull up or do I need to put a capacitor into the wiring harness?

The light on my sensor stays on the whole time there is filament inside it.  The print will pause every once in a while and I have to hit resume on the octo print interface.  I have the sensor plugged directly into the PI 3.3v, ground, and GPIO pin 2.  From reading it seems like the PI has a built in pull up or pull down resistor that can be selected with software, but I kinda got lost after reading that.

MoonshineSG, 2016-09-15:
Thats not good.... It means the switch gets false reads.... You can try to increase the bounce value and see if it helps....  

If you're using one of those endstop switches you dont need software pull (up/down). Since you mentioned there is an LED indicator on your switch, I guess you use one of theses :

[image]

I'll check the codes again. QUite difficult for me to test, as I am not actually using this plugin :D. I wrote it just as an example. And it stuck.

cbusillo, 2016-09-17:
I increased the bounce to 800 with the same results. I have the switch pictured. I might just buy the sensor you designed it for. I think you even posted a housing!

Again thanks for all the hard work, especially on something you don't use yourself.

MoonshineSG, 2016-09-18:
The housing is not by me, but thats how the plugin became a little popular... 

You might try to move to a different RPi pin, maybe a generic one (without a secondary function)

MoonshineSG, 2016-09-18:
@cbusillo I added a tiny delay in the reading to try avoiding fake reads... Try the latest code and let me know... Please check if it really pauses when filament runs out.

cbusillo, 2016-09-20:
After updating the print does not pause at all.  It does correctly show if the filament is loaded or not.

MoonshineSG, 2016-09-21:
can you paste your plugin settings and the relevant logs ?

OnAShoestring, 2016-12-28:
Hi, thanks for making and sharing this plugin.

Did you test the input signals with an oscilloscope yet? I recently made my own sensor (http://www.filamentroller.com) which *had* a 600 microsecond blip on power-on which was falsely triggering pause. I assume "bounce" was a millisecond debounce value but trying values 4 to 4000 had no effect. Am I making the wrong assumption here?

In any case, I added time.sleep(0.100) to the first line of check_status (100 ms manual software debounce) and the problem was solved.

I later modified my circuit so that my sensor no longer needs the software debounce (no more blips, very clean signal); however, I think this may be the answer to your issue here.

BETLOG, 2018-03-22:
I'm confused.
Have you added an NO/NC option yet? Surely that's fairly trivial to do.
I have the same kind of sensor as you posted in the image above. 
It's connector is wired green, null, black, red ( pins from closest to microswitch to furthest away from it).
And yet I get 0 when filament is in it, and the red led is active.... and 1 when i remove the filament and the led  is not on.
...what?

I tried moving the green signal wire from:
green, null, black, red
to:
null, green, black, red
.. I was sure that would flip NO/NC, and yet it didnt work at all. As if the switch doesnt even connect it's NC circuit or something. Would I need to reinitialise octoprint or the Pi? for it to recognise that?... Surely not.

MoonshineSG, 2018-03-22:
should work... just do a manual deployment

BETLOG, 2018-03-22:
Should work, and yet....
I've also been trying to add the simple inversion flag, but python isn't something i'm familiar with, so i'm getting frustrated by something being syntactically broken and not knowing how to debug.
Is it obvious to you where i'm breaking this?

> `nano /home/pi/.octoprint/config.yaml`

```
plugins:
  filament:
    pin: 16
    bounce: 400
    invert: 1
```

> `nano /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py`

```
        def on_after_startup(self):
                self.PIN_FILAMENT = self._settings.get(["pin"])
                self.BOUNCE = self._settings.get_int(["bounce"])
                self.INVERT = self._settings.get_int(["invert"])

                if self.PIN_FILAMENT != -1:
                        self._logger.info("Filament Sensor Plugin setup on GPIO [%s]..."%self.PIN_FILAMENT)
                        GPIO.setup(self.PIN_FILAMENT, GPIO.IN)

        def get_settings_defaults(self):
                return dict(
                        pin = -1,
                        bounce = 400,
                        invert = 0
                )

        @octoprint.plugin.BlueprintPlugin.route("/status", methods=["GET"])
        def check_status(self):
                status = "-1"
                if self.PIN_FILAMENT != -1:
                        if self.PIN_INVERT != 1:
                                status = "1" if GPIO.input(self.PIN_FILAMENT) else "0"
                        else:
                                status = "0" if GPIO.input(self.PIN_FILAMENT) else "1"
                                self._logger.info("Filament Sensor Plugin [%s] RETURNING INVERTED VALUE..."%self._identifier)
                return jsonify( status = status )

```

BETLOG, 2018-03-22:
Oh... sorry. I didn't realise this plugin was as inactive as it is...and I can see the reason.
My bad :)
It might be useful to preface the intro with an 'out of development' note to keep people like me from blundering in here expecting stuff.

Also, thanks for indirectly forcing me to look at yet another bit of python, it's one of those things I keep meaning to learn properly. :)

MoonshineSG, 2018-03-22:
most firmware now support filament sensors, so this is kinda obsolete... but it's woking, if one wants to have it working...