MoonshineSG/Octoprint-Filament #11: How to query for status?

ManuGithubSteam, 2016-09-25:
Hi

How can i query the status of the plugin. I fear it does not work.
I secured my haproxy with authentication. How can i supply this with netcat ? Is this even an issue ?

I tried to go to the haproxy (80) and octoprint directly (5000) from the pi and from my main pc.

[manuel@phobos ~]$ netcat 192.168.0.220 80 
GET /api/files HTTP/1.1
Host: 192.168.0.220
X-Api-Key: 875437E7CA1C491EA

HTTP/1.0 401 Unauthorized
Cache-Control: no-cache
Connection: close
Content-Type: text/html
WWW-Authenticate: Basic realm="octoprint"

<html><body><h1>401 Unauthorized</h1>                                                                                                                                                        
You need a valid user and password to access this content.                                                                                                                                   
</body></html>      

On the pi:
                               
tom@prusa ~ $ netcat 127.0.0.1 5000
GET /plugin/filament/status HTTP/1.1
Host: localhost
X-Api-Key: 30F7A5CFC5D540569141F74A

tom@prusa ~ $ netcat 127.0.0.1 5000

In general i miss a howto how to get that satus stuff. The doc is not very helpful if it does not work how it is described in there..... You could also provide a script who gets that, and just let the user fill in ip and api :-)

My octopint config:

filament:
    bounce: 400
    filament: 1
    pause: 1
    pin: 14
  softwareupdate:

Thanks for clearing that up :-)

ManuGithubSteam, 2016-09-25:
Found it:
curl -u USER:PASSWD --header "X-Api-Key: ????" 192.168.0.220:80/plugin/filament/status

In the config has to be: octoprint_filament

Send this command from the pi :-)

MoonshineSG, 2016-09-25:
sorry, not the setting should be "filament" ... Not sure why "octoprint_filament" works for you... (does it really?)

Can you post the logs ?

ManuGithubSteam, 2016-09-25:
Hi
When is use octoprint_filament i get:
tom@prusa ~ $ curl -u  --header "X-Api-Key: " 192.168.0.220:80/plugin/filament/status
<html><body><h1>503 Service Unavailable</h1>
No server is available to handle this request.
</body></html>
tom@prusa ~ $ curl -u --header "X-Api-Key: " 192.168.0.220:80/plugin/filament/status
{
  "status": -1
}tom@prusa ~ $ curl -u --header "X-Api-Key: " 192.168.0.220:80/plugin/filament/status
{
  "status": -1
}

If i use only filament i get: 

tom@prusa ~ $ curl -u --header "X-Api-Key: " 192.168.0.220:80/plugin/filament/status
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.</p>
tom@prusa ~ $ curl -u --header "X-Api-Key: " 192.168.0.220:80/plugin/filament/status
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.</p>
tom@prusa ~ $ curl -u --header "X-Api-Key: " 192.168.0.220:80/plugin/filament/status
<html><body><h1>503 Service Unavailable</h1>
No server is available to handle this request.
</body></html>

In the log there is this:

2016-09-25 11:40:09,959 - octoprint - ERROR - Exception on /plugin/filament/status [GET]
Traceback (most recent call last):
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 64, in check_status
    status = int(self.FILAMENT == self.CLOSED) if GPIO.input(self.PIN_FILAMENT) else int(self.FILAMENT != self.CLOSED)
RuntimeError: You must setup() the GPIO channel first
2016-09-25 11:40:14,025 - octoprint - ERROR - Exception on /plugin/filament/status [GET]
Traceback (most recent call last):
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 64, in check_status
    status = int(self.FILAMENT == self.CLOSED) if GPIO.input(self.PIN_FILAMENT) else int(self.FILAMENT != self.CLOSED)
RuntimeError: You must setup() the GPIO channel first
2016-09-25 11:40:15,030 - octoprint - ERROR - Exception on /plugin/filament/status [GET]
Traceback (most recent call last):
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/tom/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 64, in check_status
    status = int(self.FILAMENT == self.CLOSED) if GPIO.input(self.PIN_FILAMENT) else int(self.FILAMENT != self.CLOSED)
RuntimeError: You must setup() the GPIO channel first
2016-09-25 11:41:56,868 - octoprint - ERROR - Exception on /plugin/filament/status [GET]

MoonshineSG, 2016-09-26:
and your config.yaml has the entries:
```
plugins:
  filament:
   pin: XX 
``` 
 (where XX is your GPIO pin )?

post the logs from the octotprint startup ... (specifically the lines with "filament" entries)

ManuGithubSteam, 2016-09-26:
Hi
Yes my config looks like this:

octoprint_filament:
    bounce: 400
    filament: 1
    pause: 1
    pin: 14

However i used octoprint_filament because it would not work otherwise.
I can post the logs but i changed stuff in the python code (made it work with octoptint_filament too) so i could actually get it working. 

I could revert the changes and post clean logs but that has to wait till the weekend i fear...

MoonshineSG, 2016-09-26:
so you made changes to make it work with "octoprint_filament" instead of plain "filament" ?

ManuGithubSteam, 2016-09-26:
Hi ovidiu

No

I found that "filament" does not work in the config. So i use in the
config "octoprint_filament"

This however does not work in the source code so i changaged that to
work with octoprint_filament (copied the section so now the code should
work with octoprint_filament and with plain filament (but it seems not
to..))

Hope that clears it up a bit :-)

MoonshineSG, 2016-09-26:
Sorry, confused me more... can you send me your code (modified) via email ?