MoonshineSG/Octoprint-Filament #13: GPIO setup error

dragoonmc, 2016-09-25:
I am trying to get this implemented but right now the log file is giving me this log.

2016-09-25 08:46:16,308 - octoprint.plugins.filament - INFO - Running RPi.GPIO version '0.6.2'...
2016-09-25 08:46:16,309 - octoprint.plugins.filament - INFO - Filament Sensor Plugin [filament] initialized...
#other info here

2016-09-25 08:49:48,325 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2016-09-25 08:49:48,337 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 70, in on_event
    self.setup_detection()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 88, in setup_detection
    GPIO.add_event_detect(self.PIN_FILAMENT, GPIO.FALLING, callback=self.filament_runout, bouncetime=self.BOUNCE)
RuntimeError: You must setup() the GPIO channel as an input first

I don't see any other info in the log. Appears that somehow it is skipping your on_after_startup section
I am familiar with python but not good with it.

MoonshineSG, 2016-09-25:
it seams that you might not have setup the `config.yaml` ?
```
plugins:
  filament:
    pin: XX
```

dragoonmc, 2016-09-25:
it was there but i was missing the bounce setting
now that I added that I am having permission errors

2016-09-25 09:35:03,845 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 46, in on_after_startup
    GPIO.setup(self.PIN_FILAMENT, GPIO.IN)
RuntimeError: No access to /dev/mem.  Try running as root!

I have run sudo chmod...
do I need to add it to some startup file?

MoonshineSG, 2016-09-25:
no, just run once..

dragoonmc, 2016-09-25:
for some reason it does not seem to want to maintain those permissions

dragoonmc, 2016-09-25:
got past permission errors by adding a command to /etc/rc.local

dragoonmc, 2016-09-25:
Now I get this error

2016-09-25 10:45:04,964 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2016-09-25 10:45:05,993 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 70, in on_event
    self.setup_detection()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 88, in setup_detection
    GPIO.add_event_detect(self.PIN_FILAMENT, GPIO.FALLING, callback=self.filament_runout, bouncetime=self.BOUNCE)
RuntimeError: Failed to add edge detection

MoonshineSG, 2016-09-26:
I still can't see the  `Filament Sensor Plugin setup on GPIO ...` line which indicates the plugin is properly setup... 

(btw, the bounce setting is not important -  yet. there are default values if you don't specify one )

dragoonmc, 2016-09-26:
Sorry I was only posting part of the log. Here is everything from startup. I did add one line to verify all setting in config.yaml are being set correctly in the plugin

2016-09-25 13:53:11,985 - octoprint.server - INFO - Starting OctoPrint 1.2.16 (master branch)
2016-09-25 13:53:12,040 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-09-25 13:53:12,845 - octoprint.plugin.core - INFO - Found 9 plugin(s) providing 8 mixin implementations, 4 hook handlers
2016-09-25 13:53:12,854 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-09-25 13:53:14,292 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-09-25 13:53:14,374 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2016-09-25 13:53:15,877 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-09-25 13:53:15,881 - octoprint.plugins.filament - INFO - Running RPi.GPIO version '0.6.2'...
2016-09-25 13:53:15,882 - octoprint.plugins.filament - INFO - Filament Sensor Plugin [filament] initialized...
2016-09-25 13:53:15,884 - octoprint.plugin.core - INFO - Initialized 8 plugin implementation(s)
2016-09-25 13:53:15,891 - octoprint.plugin.core - INFO - 9 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
|  EEPROM Printrbot variation of Marlin Editor Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin_printrbot
|  Filament-Sensor (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament
|  Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/softwareupdate
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
2016-09-25 13:53:15,893 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-09-25 13:53:15,909 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-09-25 13:53:15,911 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-09-25 13:53:15,927 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-09-25 13:53:16,860 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2016-09-25 13:53:16,902 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-09-25 13:53:16,917 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-09-25 13:53:16,921 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2016-09-25 13:53:16,922 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-09-25 13:53:17,196 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup indicates '1' when filament is present (setup on GPIO 4).
2016-09-25 13:53:17,199 - octoprint.plugins.filament - INFO - Current status is '1'...
2016-09-25 13:53:17,200 - octoprint.plugins.filament - INFO - Other settings are Bounce '400' & pause '0'.
2016-09-25 13:53:17,259 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.178.1.34/'}
2016-09-25 13:53:20,962 - octoprint.server.util.sockjs - INFO - New connection from client: 192.178.1.168
2016-09-25 13:54:04,969 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2016-09-25 13:54:06,005 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 71, in on_event
    self.setup_detection()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 89, in setup_detection
    GPIO.add_event_detect(4, GPIO.FALLING, callback=self.filamentrunout, bouncetime=400)
RuntimeError: Failed to add edge detection

MoonshineSG, 2016-09-26:
you're using the latest code not the one one the latest release, right ? 
that still has some bugs... it seams that I can't use both FALLING and RISING so I need to find another way to support both.... but I didn't get around to doing it... 

I suggest to switch to the release version... (for now)

dragoonmc, 2016-09-26:
I am using the code from here. Is the latest lease the one on octoprints site?

MoonshineSG, 2016-09-26:
"here" are multiple version... so depends how do you install it in octoprint... via
`https://github.com/MoonshineSG/Octoprint-Filament/archive/master.zip` (manually) or
`https://github.com/MoonshineSG/Octoprint-Filament/archive/1.0.1.zip` (using the OctoPrint plugin repository)

dragoonmc, 2016-09-26:
octoprint repository is looking at master

MoonshineSG, 2016-09-26:
damn.. you're right. 
working on a fix... give me a while to test it...

dragoonmc, 2016-09-26:
btw it is failing to even set the first edge detect for me

MoonshineSG, 2016-09-26:
i changed some codes... not fully tested, but you can try it...

dragoonmc, 2016-09-26:
will do as soon as i figure out how to uninstall a plugin that the manager isn't seeing

MoonshineSG, 2016-09-26:
why the manager can't see it ? 

hold on. I am splitting the codes to a new devel branch... it's messed up ...

dragoonmc, 2016-09-26:
I don't know, I did install manually from a zip before i realized it was the same as the repository right now. but the manager is not showing it

MoonshineSG, 2016-09-26:
you need to remove 2 folders:

```
/home/pi/oprint/lib/python2.7/site-packages/Filament_Sensor-1.0.1-py2.7.egg-info
/home/pi/oprint/lib/python2.7/site-packages/octoprint_filament
```

MoonshineSG, 2016-09-26:
then to install the devel codes, run 

`pip install https://github.com/MoonshineSG/OctoPrint-Filament/archive/devel.zip`

dragoonmc, 2016-09-26:
Still the same issue for me, maybe the problem is somewhere else in my Pi. I will try a fresh version after work tomorrow

2016-09-25 22:30:57,076 - octoprint.server - INFO - Starting OctoPrint 1.2.16 (master branch)
2016-09-25 22:30:57,129 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-09-25 22:30:57,907 - octoprint.plugin.core - INFO - Found 9 plugin(s) providing 8 mixin implementations, 4 hook handlers
2016-09-25 22:30:57,915 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-09-25 22:30:59,276 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-09-25 22:30:59,353 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2016-09-25 22:31:00,836 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-09-25 22:31:00,840 - octoprint.plugins.filament - INFO - Running RPi.GPIO version '0.6.2'...
2016-09-25 22:31:00,841 - octoprint.plugins.filament - INFO - Filament Sensor Plugin [filament] initialized...
2016-09-25 22:31:00,844 - octoprint.plugin.core - INFO - Initialized 8 plugin implementation(s)
2016-09-25 22:31:00,850 - octoprint.plugin.core - INFO - 9 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
|  EEPROM Printrbot variation of Marlin Editor Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin_printrbot
|  Filament-Sensor (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament
|  Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/softwareupdate
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
2016-09-25 22:31:00,852 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-09-25 22:31:00,867 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-09-25 22:31:00,871 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-09-25 22:31:00,886 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-09-25 22:31:01,999 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2016-09-25 22:31:02,042 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-09-25 22:31:02,059 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-09-25 22:31:02,064 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2016-09-25 22:31:02,067 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-09-25 22:31:02,078 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup indicates '1' when filament is present (setup on GPIO 4).
2016-09-25 22:31:02,080 - octoprint.plugins.filament - INFO - Current status is '1'...
2016-09-25 22:31:02,392 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.178.1.34/'}
2016-09-25 22:31:05,946 - octoprint.server.util.sockjs - INFO - New connection from client: 192.178.1.168
2016-09-25 22:31:13,955 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2016-09-25 22:31:14,977 - octoprint.plugin - ERROR - Error while calling plugin filament
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 70, in on_event
    self.setup_detection()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/__init__.py", line 84, in setup_detection
    GPIO.add_event_detect(self.PIN_FILAMENT, GPIO.BOTH, callback=self.filament_detection, bouncetime=self.BOUNCE)
RuntimeError: Failed to add edge detection

MoonshineSG, 2016-09-26:
maybe. you have problems with the permission as well... Maybe a fresh install will help. 
let me know how it goes.

dragoonmc, 2016-09-27:
Got a new download of the latest Octopi, ran all Apt-get update, upgrade, and dist-upgrade
all fine now with current master version. 
I should've just done that in the first place sorry for bothering you

MoonshineSG, 2016-09-27:
no bother at all. i killed a big bug and reorganized my source code.. 
i'm glad it works for you now. 

using master or devel ?

dragoonmc, 2016-09-27:
master from about an hour ago

dragoonmc, 2016-09-27:
Automatic resume does not work but I am not having any errors and pausing is working

MoonshineSG, 2016-09-27:
No, automatic resume is not implemented.  I leave it to the user to resume when everything is ready.  There might be additional steps needed when new filament is added.

dragoonmc, 2016-09-27:
What is the point of the rising edge detect then in my case

MoonshineSG, 2016-09-27:
it triggers an event and for other plugins to be notified. other people can implement whatever they want in that plugin. Thats the reason why there a "pause = 0"  settings where this plugin does not pause the print, but only raise an event. other plugins can implement actions for the 2 events. 

But that is in the devel version..

dragoonmc, 2016-09-27:
ah ok makes more sense then.
thanks for the plugin. Makes life so much easier to have it stop when it runs out of filament for some reason.

MoonshineSG, 2016-09-27:
👍 
Maybe one day I'll make use of it as well..... :D

trgtylcnky, 2017-03-27:
Hi, I have the same problem: RuntimeError: Failed to add edge detection
I tried a python script that adds event, the same error. But when I run the script with sudo, it works OK. I added pi user to gpio group but no luck.

dragoonmc, 2017-03-27:
I ended up getting it to work just fine. Don't remember how I got it to work

closed 2017-05-14 by dragoonmc