MoonshineSG/Octoprint-Filament #14: Showing 'present' regardless of filament status

unbwogable, 2016-10-01:
I've confirmed my switch is normally open, verified it's working, tried a few pins, but I'm consistently getting a response that the filament is preset/1, regardless of if it is or not.

MoonshineSG, 2016-10-01:
Post your config and the logs.

unbwogable, 2016-10-01:
config:
plugins:
  _disabled:
  - cura
  - detailedprogress
  - discovery
  - virtual_printer
  announcements:
    channels:
      _important:
        read_until: 1460575175
  filament:
    bounce: 400
    pin: 10
  softwareupdate:
    _config_version: 4
    checks:
      octoprint:
        checkout_folder: /home/pi/OctoPrint
        prerelease: false
  touchui:
    hasCustom: false
    refreshCSS: false
    requireNewCSS: false
    whatsNew: false

log:

2016-09-29 03:55:15,535 - octoprint.server.util.sockjs - INFO - New connection from client: [my ip]
2016-09-29 05:11:42,477 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.16
2016-09-29 05:16:01,605 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cage_for_testing.gcode
2016-09-29 05:29:16,366 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.16
2016-09-29 05:30:49,026 - octoprint.server.util.sockjs - INFO - Client connection closed: [my ip]
2016-09-29 05:32:57,513 - octoprint.server.util.sockjs - INFO - New connection from client: [my ip]
2016-09-29 05:37:37,559 - octoprint.server.util.sockjs - INFO - Client connection closed: [my ip]
2016-09-29 07:08:44,601 - octoprint.server.util.sockjs - INFO - New connection from client: [my ip]
2016-09-29 07:13:40,300 - octoprint.server.util.sockjs - INFO - Client connection closed: [my ip]
2016-09-29 07:14:01,416 - octoprint.server.util.sockjs - INFO - New connection from client: [my ip]
2016-09-29 07:20:00,435 - octoprint.server.util.sockjs - INFO - Client connection closed: [my ip]
2016-09-29 15:13:35,991 - octoprint.server.util.sockjs - INFO - New connection from client: [my ip]
2016-09-29 15:21:00,371 - octoprint.server.util.sockjs - INFO - Client connection closed: [my ip]
2016-09-29 16:13:01,752 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.27
2016-09-29 16:37:06,940 - octoprint.server.util.sockjs - INFO - New connection from client: [my ip]
2016-09-29 16:54:39,453 - octoprint.server.util.sockjs - INFO - Client connection closed: [my ip]
2016-09-29 17:25:45,706 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cage_for_testing.gcode
2016-09-29 21:17:11,638 - octoprint.server - INFO - Starting OctoPrint 1.2.16 (master branch)
2016-09-29 21:17:11,693 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-09-29 21:17:12,923 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 5 mixin implementations, 3 hook handlers
2016-09-29 21:17:12,932 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-09-29 21:17:13,171 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-09-29 21:17:13,230 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2016-09-29 21:17:18,161 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 8.1.1
2016-09-29 21:17:18,166 - octoprint.plugin.core - INFO - Initialized 5 plugin implementation(s)
2016-09-29 21:17:18,173 - octoprint.plugin.core - INFO - 10 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
| !Detailed Progress Plugin (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
| !Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
| !Filament-Sensor (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
|  Print History Plugin (0.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/softwareupdate
|  STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
2016-09-29 21:17:18,178 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-09-29 21:17:18,193 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-09-29 21:17:18,198 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-09-29 21:17:18,224 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-09-29 21:17:19,218 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2016-09-29 21:17:19,305 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
[... 1744 lines cut ...]
2016-10-01 00:33:09,212 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://47.217.98.86:5000/'}
[... cut ...]
2016-10-01 00:37:02,876 - octoprint.plugin.core - INFO - 12 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
| !Detailed Progress Plugin (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
| !Discovery (bundled) = /home/pi/op

MoonshineSG, 2016-10-01:
I see that the plugin detects the status at the beginning,  but can't see any calls to the api...

```
2016-10-01 00:19:38,205 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup indicates '0' when filament is present (setup on GPIO 7).
2016-10-01 00:19:38,207 - octoprint.plugins.filament - INFO - Current status is '1'...

2016-10-01 00:28:38,355 - octoprint.plugins.filament - INFO - Filament Sensor Plugin setup indicates '0' when filament is present (setup on GPIO 7).
2016-10-01 00:28:38,356 - octoprint.plugins.filament - INFO - Current status is '0'...

```

which version of the plugin are you using ? master or devel ?

unbwogable, 2016-10-01:
Should be master, I removed and reinstalled to be sure

MoonshineSG, 2016-10-01:
btw, the detection only happens during printing.. :D 

if you use the devel, I just updated to it always detects (makes more sense for this version)