MoonshineSG/Octoprint-Filament #18: Sensor default status

murathanaraz, 2017-03-04:
https://ae01.alicdn.com/kf/HTB1jVYgMXXXXXX9XXXXq6xXFXXXj/5pcs-Robot-Model-car-font-b-helicopter-b-font-font-b-Crash-b-font-collision-switch.jpg

I am using this endstop sensor but couldn't get success. When I unplug the sensor octopi pauses and it wont resume with a click until I plug it back. Which is a good thing. But the sensor is not working as expected. Once I made it work but it works random. Not reliable. 

Maybe the default status of my sensor is the problem, I have connected OUT to GPIO 17, VCC to 5v, Ground to Ground.

Any suggestions?

MoonshineSG, 2017-03-05:
what was this all about ? :D

murathanaraz, 2017-03-05:
I was asking how to add gcode when filament runs out. Maybe a M600 command for filament unload and wait for the load while beeping. But ı was not sure if this is the right place to ask😃

MoonshineSG, 2017-03-05:
oh... for that you might wana try the "devel" branch. that has configurable codes. but you need to manually install it.