MoonshineSG/Octoprint-Filament #23: Sensor state detected but no pause

uwe999, 2017-05-11:
I'm having same issue as https://github.com/MoonshineSG/Octoprint-Filament/issues/20

logs show:
2017-05-11 22:39:27,680 - octoprint.plugins.filament - DEBUG - Detected sensor [26] state [1]? !
2017-05-11 22:39:29,481 - octoprint.plugins.filament - DEBUG - Detected sensor [26] state [0]? !
2017-05-11 22:39:29,482 - octoprint.plugins.filament - DEBUG - Sensor [0]!

Printer isn't pausing. Any ideas why?

thanks
UWE

MoonshineSG, 2017-05-12:
no, not from the information provided.
the referred issue #20 was closed by the user without any explanation. 

maybe you can provide more details... 
- printer model ?
- printing from SD or local ?
- have you tried to pause a print, make sure the octoprint can pause it ?

anything thats "special" about your printer/model... can help

uwe999, 2017-05-12:
thanks for the quick reply!
I have a Wanhao Duplicator i3 clone (Monoprice Maker Select V2.1), current 1.3.2 version of Octoprint, printing from a file uploaded to Octoprint. I can pause a print in Octoprint without an issue.

Can you tell me where the __init__.py file is located on my pi, then I could add some more debugging to the code.

Thanks again
UWE

MoonshineSG, 2017-05-12:
it should be at `/home/pi/site-packages/octoprint_filament`

uwe999, 2017-05-12:
sorry, no such directory

MoonshineSG, 2017-05-12:
Apologies. That's my shortcut...
Should be "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filament/"