MoonshineSG/Octoprint-Filament #24: New feature: Notification after runout

trgtylcnky, 2017-06-07:
Hi,
I think it would be nice to see a notification when filament runout is detected.

MoonshineSG, 2017-06-07:
Yes it would be. And it will be... :) IF you have Nautilus installed, the new release (in a few days) will come with notification support. And I'll add filament notification.

trgtylcnky, 2017-06-07:
Hmm, I am actually talking about octoprint web interface not the mobile device. Since you mentioned Nautilus, I am a little confused.

MoonshineSG, 2017-06-07:
oh.... the web interface would have a "paused" notification... (i guess)

sorry if i confused you with Nautilus...

MoonshineSG, 2017-06-07:
also, the plugin triggers and event that can be used by other plugins to show any kind of notifications...

Actpohomoc, 2017-06-07:
I'm using home-assistant automation notification with telegram notify
component. It is possible now to send even picture from webcam with
notification message.
Also I've setup the automation notification for start and finish the job if
the 3d printer.