MoonshineSG/Octoprint-Filament #29: i need to modify this plugin to read a gpio pin (check voltage ) and if power goes off pause the print

amrithmmh, 2018-03-30:
so what changes should i make to just pause the print if power goes off(i will be running raspberry pi on battery if no power ) and just get the current x y z value and save it ?

MoonshineSG, 2018-03-30:
The plugin does just that. Reads a gpio pin and act on the change (drop or raise, depending on the configuration)

amrithmmh, 2018-03-30:
@MoonshineSG  but it checks for some sensor ?

MoonshineSG, 2018-03-30:
Hmmm... it read the status of the pin... it’s either “up” or “down”
I am not sure I understand what you are trying to do

amrithmmh, 2018-03-30:
@MoonshineSG  i need to read voltage using raspberry pi running Octopi (which will have battery backup).
when power goes off the print will stop but the raspberry pi will still be powered and will give power to arduino for 3d printer (but not the motors). So i just want to check volatge level with 12V DC using raspberry and if power goes off the reading goes to 0V and i need to pause the print and save current x y z direction (to any file say abc.txt)....i want to later read the values and start printing but i want to implement just upto saving the file part

MoonshineSG, 2018-03-30:
So you will have both, 12V and battery connected to you RPi ? Not sure that’s a good idea. 
But if you do, you need to find a way to tell the plugin which power source is active (basically just know if 12V is NOT active)

One way I can think of is to have a voltage drop from 12 to 5 on one of the GPIo and let the plugin read that... This is just an idea and needs to be tested if it really works

amrithmmh, 2018-03-30:
i think this is the main part i should look?

	def check_gpio(self, channel):
		state = GPIO.input(self.PIN_FILAMENT)
		self._logger.debug("Detected sensor [%s] state [%s]? !"%(channel, state))
		if not state: #safety pin ?
			self._logger.debug("Sensor [%s]!"%state)
			if self._printer.is_printing():
				self._printer.toggle_pause_print()

amrithmmh, 2018-03-30:
channel is GPIO pin and state is 0 or 1 right?

amrithmmh, 2018-03-30:
# coding=utf-8
from __future__ import absolute_import

import octoprint.plugin
import octoprint.settings
import octoprint.util

from octoprint.events import eventManager, Events
from flask import jsonify, request

import logging
import logging.handlers
import RPi.GPIO as GPIO 

class Powerpo(octoprint.plugin.StartupPlugin,
							octoprint.plugin.SettingsPlugin,
							octoprint.plugin.EventHandlerPlugin,
							octoprint.plugin.BlueprintPlugin):

	def initialize(self):
		self._logger.setLevel(logging.DEBUG)
		
		self._logger.info("Running RPi.GPIO version '{0}'...".format(GPIO.VERSION))
		if GPIO.VERSION < "0.6":
			raise Exception("RPi.GPIO must be greater than 0.6")
			
		GPIO.setmode(GPIO.BCM)
		GPIO.setwarnings(False)
		
		self._logger.info("power pause [%s] initialized..."%self._identifier)

	def on_after_startup(self):
		self.PIN_POWER = self._settings.get(["pin"])
		
		if self.PIN_POWER != -1:
			self._logger.info("Power pause Plugin setup on GPIO [%s]..."%self.PIN_POWER)
			GPIO.setup(self.PIN_POWER, GPIO.IN)
		
	def get_settings_defaults(self):
		return dict(pin = -1)

	@octoprint.plugin.BlueprintPlugin.route("/status", methods=["GET"])
	def check_status(self):
		status = "-1"
		if self.PIN_POWER != -1:
			status = "1" if GPIO.input(self.PIN_POWER) else "0"
		return jsonify( status = status )
		
     
	def setup_gpio(self):
		try:
			GPIO.remove_event_detect(self.PIN_POWER)
		except:
			pass
		if self.PIN_POWER != -1:
			GPIO.add_event_detect(self.PIN_POWER, GPIO.FALLING, callback=self.check_gpio) 

	def check_gpio(self, channel):
		state = GPIO.input(self.PIN_POWER)
		self._logger.debug("Detected sensor [%s] state [%s]? !"%(channel, state))
		if not state: #safety pin ?
			self._logger.debug("Sensor [%s]!"%state)
			if self._printer.is_printing():
				self._printer.toggle_pause_print()

__plugin_name__ = "Power-pause"
__plugin_version__ = "1.0.1"
__plugin_description__ = "Use a voltage measurement to pause printing when power goes out."

def __plugin_load__():
	global __plugin_implementation__
	__plugin_implementation__ = Powerpo()

amrithmmh, 2018-03-30:
how to use the above .py file as plugin?
are there any errors?
the above code just checks a GPIO and pauses print

MoonshineSG, 2018-03-30:
sorry, not sure what you're asking.... if you modify the plugin to do a completely different thing, I can only assume you know what you're doing... 
There is no way for me to look at your codes and find bugs and tell you what can go wrong.

Creating a plugin without python knowledge is a task few would succeed at. 

If your python skills are decent, then maybe you can start reading about how to develop an OctoPrint plugin at http://docs.octoprint.org/en/master/plugins/

amrithmmh, 2018-03-31:
@MoonshineSG i do know python ...thank you for that link i was looking for the same thing...thank you

amrithmmh, 2018-04-03:
@MoonshineSG  i almost got it working but i don have some questions if you could help me that would be great 
 GPIO.FALLING is used to check if there is state change from high to low
will the printing not pause if the state is 0 from the beginning of the print?

amrithmmh, 2018-04-03:
i have tested and it works only if there is a change and not by default...so your plugin will not pause if the filament is not present from the beginning of the print i guess

MoonshineSG, 2018-04-03:
That’s correct. I assume that when you start a print you visually check the presence of filament. The plugin deals with events that happen while you are most likely away from the printer

amrithmmh, 2018-04-03:
@MoonshineSG it will be great if you could check first for filament right after giving print...(believe me there are people who are too lazy even to check before printing)  ...just a suggestion thyough

thank you