MoonshineSG/Octoprint-Filament #30: It does nothing...

fr2632, 2018-04-16:
When filament is running out, my sensor just beeps and I get error "abnormal sensor" on the LCD. The extruder keeps going without pause.

MoonshineSG, 2018-04-17:
i'm not sure how the LCD comes in the picture. The plugin does not handle any LCD display. That sounds more like the firmware reading the sensor not the octoprint ....

Are you sure the plugin is configured correctly ?

fr2632, 2018-04-17:
Well, the plugin features says : "Displays detailed progress on the LCD screen" but I dont see anything happen on the LCD screen. This picture was taken while printing.

Take a look here : https://plugins.octoprint.org/plugins/detailedprogress/

MoonshineSG, 2018-04-17:
it's all in the plugin description. please read it.