MoonshineSG/Octoprint-Filament #31: works only with external pull-up or pull-down resistor

m-stanko, 2019-02-21:
May I suggest additional changes to existing code and additional setting in config.yaml?
config.yaml (parameter logic, 0 or 1, holds default logic value of sensor pin):
	logic : 1 

in __init__.py:
line 34: 
 		self.LOGIC = self._settings.get_int(["logic"])
line 38-40:
			if self.LOGIC == 1:
				GPIO.setup(self.PIN_FILAMENT, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
			else:
				GPIO.setup(self.PIN_FILAMENT, GPIO.IN, pull_up_down=GPIO.PUD_UP)
line 75:
			GPIO.add_event_detect(self.PIN_FILAMENT, GPIO.BOTH, callback=self.check_gpio, bouncetime=self.BOUNCE) 

line 80:
		if state != self.LOGIC : #safety pin ?

Story background: If you are able to set default pin value during GPIO.setup You may set internal pull-up or pull-down resistor. Without pull-up/dn resistor input pin will be somewhere "in the middle" state and may cause false falling/rising event

m-stanko, 2019-02-21:
i forgot lines 44-49 with additional dictionary member:
	def get_settings_defaults(self):
		return dict(
			pin = -1,
			bounce = 300,
			logic = 1
		)

MoonshineSG, 2019-02-22:
Glad yo find the plugin usefull but I don't really maintain this anymore.  Someone took it and improved it greatly, so you migth want to have a look at https://github.com/kontakt/Octoprint-Filament-Reloaded

closed 2019-02-22 by MoonshineSG

m-stanko, 2019-02-22:
Well, thats fine by me. If there is no update my local modifications won't be overwritten. Anyway I want to thank You for your contribution which is quite useful to me.
BR